dwhinham/mt32-pi #338: Power saving mode isn't work ob RPi 3A+ 🐛 [BUG]

yarodin, 2023-04-18:
## Hardware and software

- **mt32-pi version:** _v0.13.1_
- **Raspberry Pi model:** _Raspberry Pi 3 Model A+_
- **HATs:** _e.g. Pi-MIDI_
- **USB devices:** _mt32-pi-midi-hat_
- **MIDI host:** _MisTer_

## Bug description

Looks like power saving mode isn't setting after the timeout. OLEDs backlight is on always.

## Steps to reproduce

1. Power on mister with connected mt32-pi-midi-hat.
2. After certain number of second of inactivty (power_save_timeout = 300) OLEDs backlight is still on.

## Expected behavior

After 300 seconds of inactivity OLED backlight will off and RPi goes to power saving mode.

## Configuration file

<details>
<summary>Click to expand</summary>

```ini
#                                  ________    _______                      __
#                         __      /_____   `. /____   `.                   /__`
#     ____   ____     __/  /____  _______)  / ______)  / ____  ______      __
#   /   __ v  __  `./__   _____//_____    < /   _____. /____//   ___  `. /  /
#  /  /  /  /  /  /   /  /____  _______)  //  /______       /  /____/  //  .__
# /__/  /__/  /__/    \______//_________. /_________/      /   ______.  \____/
# /////////////////////////////////////////////////////// /  / //// /// // /
#                                                         ```
# mt32-pi.cfg: mt32-pi configuration file.
# Default options are marked with an asterisk (*).

# -----------------------------------------------------------------------------
# System options
# -----------------------------------------------------------------------------
[system]

# Enable or disable verbose startup and error output.
#
# When enabled, outputs more information to the LCD when starting up, and when
# MIDI/UART errors are detected.
#
# This also may hide the boot logo on smaller graphical displays.
#
# Values: on, off*
verbose = off

# Set the default synthesizer to be made active on startup.
#
# If the default synthesizer is unavailable (e.g. missing ROMs or SoundFonts),
# the first working synth is made active.
#
# Values: mt32*, soundfont
#
# mt32:      Use mt32emu (Munt) for Roland MT-32 emulation
# soundfont: Use FluidSynth for SoundFont synthesis
default_synth = mt32

# Enable or disable support for USB devices.
#
# Disable this to speed up boot time if you are not using any USB devices.
#
# Values: on*, off
usb = off
[... 415 lines cut ...]
# Compatible with MiSTer MidiLink.
#
# Values: on*, off
udp_midi = off

# Enable or disable the embedded FTP server.
#
# This FTP server is a very basic implementation which DOES NOT feature any kind
# of transport layer security/encryption. Therefore, you should NOT enable this
# feature on a public network or expose the Raspberry Pi to the Internet.
#
# The FTP server should be considered a convenience feature only, for performing
# updates and configuration changes without having to replace the SD card.
#
# Values: on, off*
ftp = on

# Set the FTP server username and password.
#
# Values: any ASCII string (mt32-pi*)
ftp_username = mt32-pi
ftp_password = mt32-pi

```
</details>

dwhinham, 2023-04-18:
OLED model? Details of who built the HAT and where you got it from?

OLEDs do not have a backlight, so please clarify what exactly you are seeing. The expected behaviour for the supported OLEDs is for the screen to simply turn blank.

If it doesn't work with a certain screen manufacturer then you may just have to disable/ignore this feature.

yarodin, 2023-04-18:
Yes, sorry, OLED wich do not have a backlight and expected behaviour for my OLED is the screen to turn blank.
I did some more investigations and find out that: 

1. If after power on I starting cores without mt32-pi support  or even do nothing, mt32-pi will never goes to power saving mode (OLED isn't turn blank)

2. If I at least one time starting core with mt32-pi support (ao486 in my case) and application wich using MIDI (DOS monkey island for example) then the mt32-pi behaviouring normally until power off - after each 300 seconds of inactivity goes to power saving mode (OLED turn blank)  

I don't know details about my version of mt32-pi. I got completed device at russians online marketplace. 
[image]
[image]
[image]

dwhinham, 2023-04-24:
Please test this build: https://github.com/dwhinham/mt32-pi/actions/runs/4790017818

I only found one bug with sleep mode, which was when a network connection or re-connection notification (e.g. WiFi) could cause the screen to display a message and then force it to stay turned on.

Otherwise, everything else related to power saving mode works correctly with my MiSTer on official hardware, on all the cores I tested.

yarodin, 2023-04-24:
I have tested this build and now power saving mode is working correctly in all possible cases. Thanks a lot.

closed 2023-04-24 by dwhinham