dwhinham/mt32-pi #344: 🐛 [BUG]

K-rnivoro, 2023-06-11:
There is this strange column of garbage to the left of the screen. I've tried other displays modules with the same result. Parameters are `type = sh1106_i2c, width = 128, height = 64, i2c_lcd_address = 3c` for the Diymore display. I'm pretty sure it worked okay with previous MT32_Pi versions.
Sorry if it is not a bug, but I've tried everything.

[image]

GoodSpeed001, 2023-06-12:
Hi @K-rnivoro ,

I have the same screen, please confirm your settings, I have the following:

`# -----------------------------------------------------------------------------
# LCD/OLED display options
# -----------------------------------------------------------------------------
[lcd]

# Select LCD driver.
#
# Note that LCDs connected via I2C, you must also set the correct address for
# your device via the i2c_lcd_address option. Consult its datasheet, or see
# our documentation for tested models and their configurations.
#
# Values: none*, hd44780_4bit, hd44780_i2c, sh1106_i2c, ssd1306_i2c
#
# none:         No LCD
# hd44780_4bit: Hitachi HD44780 or compatible (e.g. WS0010, RS0010) character
#               LCD connected to GPIO pins in 4-bit mode (see documentation for
#               pinout)
# hd44780_i2c:  As above, but using an I2C "backpack"
# sh1106_i2c:   Small I2C-based OLED graphical display (usually 1.3")
# ssd1306_i2c:  Small I2C-based OLED graphical display (usually 0.96")
type = ssd1306_i2c

# Set the width of the LCD.
#
# If the display is a character display, this value is measured in characters.
# Otherwise, for a graphical display, this is measured in pixels.
#
# Note that not all dimension settings are valid; see documentation for valid
# configurations for each LCD driver.
#
# If set to 132, assume this display is an SSD1305 (see wiki).
# Note: the correct settings for the SSD1305 may change in future.
#
# Values: 20-132 (128*)
width = 128

# Set the height of the LCD.
#
# Same characters vs. pixels considerations as for width.
#
# Values: 2-64 (32*)
height = 64

# Set address (hexadecimal) of I2C LCD.
#
# This will be used to communicate with LCDs connected via the I2C bus.
#
# Values: 00-80 (3c*)
i2c_lcd_address = 3c

# Rotate the display output (graphical LCDs only).
#
# Some graphical displays support rotation. Use this option if you need to turn
# the display around.
#
# Values: normal*, inverted
#
# normal:   No rotation
# inverted: The display output is upside down
rotation = normal

# Mirror the display output horizontally (graphical LCDs only).
#
# Some graphical displays display columns right-to-left rather than left-to-
# right. Use this option if you need to mirror the display horizontally.
#
# Values: normal*, mirrored
#
# normal:   No mirroring
# mirrored: The display output is mirrored horizontally
mirror = normal`

If that didn't work, try different type of screen, otherwise, that screen might be damaged.

hope it helps,

Regards,
GS

GoodSpeed001, 2023-06-12:
some sellers of oled screens advertise the screens like sh1106_i2c, but actually they are ssd1306_i2c

K-rnivoro, 2023-06-13:
You were right about the `type`. It is a `ssd1306`, not a `sh1106`. Now works flawlessly. I got confused by the links in the Wiki. Thanks @GoodSpeed001 !!!!

closed 2023-06-13 by K-rnivoro